litex-hub/zephyr-on-litex-vexriscv #1: third_party/litex_boards is pinned at a too-old version

tcal-x, 2020-08-11:
I get an error because `platforms/arty.py` doesn't have `_i2s_pmod_io` at this older commit, `1356ebb4`.  

If I check out current `master` for `third_party/litex_boards`, this issue is resolved (but then I encounter another issue).

tcal-x, 2020-08-11:
The new issue that I saw was due to `request_all`, used by current master targets/arty.py.   Its definition was recently added to litex/litex/build/generic_platform.py, so it's not in the version pinned here.   I think I can work around the issue by checking out litex-boards before the commit that uses `request_all`.

Ok, it seems to work if I use commit 8583c440 for litex-boards, and leave all of the other third party repos pinned where they are.

tcal-x, 2020-08-11:
FYI @pawelsag @mateusz-holenko

mateusz-holenko, 2020-10-20:
@tcal-x are you sure you are using the current versin of the repo? After 87965b053fca3b0b95a300cf43122bbd8f194b13 `litex_boards` should be ok - we couldn't reproduce the problem.

tcal-x, 2020-10-20:
Thanks @mateusz-holenko , I hope to give it another try before our Wednesday meeting.

tcal-x, 2020-12-13:
Yes it does work for me now, thanks all.

closed 2020-12-13 by tcal-x