litex-hub/zephyr-on-litex-vexriscv #2: make.py --load is broken -- no ./prog/openocd_xilinx.cfg

tcal-x, 2020-08-12:
This can probably be fixed easily, just by adding a prog/ directory to this repo, or, use the config files loaded with third_party/litex-boards.

mateusz-holenko, 2020-10-20:
@tcal-x in e4b4f99 we refactored the code and reimplemented biststream loading logic. Please check if the current version works fine for you.

tcal-x, 2020-10-21:
On master, after building, when I run `./make.py --board=arty --load`, I get:

```
python3 -m litex.soc.software.memusage bios.elf /home/tim/litex-hub/zephyr-on-litex-vexriscv/build/arty/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 23.62KiB     (37.79%)
RAM usage: 1.63KiB      (20.41%)

make: Leaving directory '/home/tim/litex-hub/zephyr-on-litex-vexriscv/build/arty/software/bios'
Traceback (most recent call last):
  File "./make.py", line 96, in <module>
    main()
  File "./make.py", line 93, in main
    board.load(soc, filename=os.path.join(build_dir, "gateware", "arty" + board.bitstream_ext))
  File "./make.py", line 20, in load
    prog.load_bitstream(filename)
  File "/home/tim/litex-hub/zephyr-on-litex-vexriscv/third_party/litex/litex/build/openocd.py", line 23, in load_bitstream
    config = self.find_config()
  File "/home/tim/litex-hub/zephyr-on-litex-vexriscv/third_party/litex/litex/build/generic_programmer.py", line 71, in find_config
    import requests
ModuleNotFoundError: No module named 'requests'
```

tcal-x, 2020-12-13:
Update: I figured out the "No module named `requests`" error -- that was from accidentally getting Python3.6 in my path.  

However, the "--load" option still does not work (if make.py is invoked directly).   Yes, the bitstream is copied to ./binaries/ now, and the user can use openocd, but if they don't already have it aliased, it's a bit of a pain to go find the xc7 / ftdi 2232 cfg file.

Was "--load" expected to work after the commit mentioned above?   There is no ./prog/openocd_xilinx.cfg.