litex-hub/zephyr-on-litex-vexriscv #3: UART communication broken with on upstream LiteX+Zephyr

lindemer, 2020-12-16:
I've experienced two separate but possibly related bugs attempting to run Zephyr on LiteX.

1. Full description here: https://github.com/zephyrproject-rtos/zephyr/issues/30702. After `zephyr-v2.1.0` the Zephyr shell module no longer accepts keyboard input.
2. If I update the LiteX sources in this repository by running `./litex_setup.py init install --user` from the `third_party/litex` directory, the problem gets worse. The Zephyr shell is broken on `zephyr-v2.1.0` and even the `hello_world` sample fails in releases more recent that this. 

I haven't had any luck tracking down the source of these bugs, and I'm unable to proceed with development until this is fixed.

tgorochowik, 2021-01-08:
Hi @lindemer 
The issue should be fixed by https://github.com/litex-hub/zephyr-on-litex-vexriscv/commit/e94a239c9396debdbbb875b0d75d615c20308552 (thanks to moving the peripherals around a little bit).

lindemer, 2021-01-08:
Thanks @tgorochowik, it works on my end.

closed 2021-01-08 by lindemer