litex-hub/zephyr-on-litex-vexriscv #5: Please update LiteX sources to include "Secure" VexRiscv cores

lindemer, 2021-01-08:
I'd like to run Zephyr on the new PMP-enabled VexRiscv cores added to the LiteX sources here https://github.com/litex-hub/pythondata-cpu-vexriscv/pull/9. I was able to build and load with `VexRiscv_Secure` by making the following changes in this repository:

- Add `cpu_variant="secure",` below this line https://github.com/litex-hub/zephyr-on-litex-vexriscv/blob/e94a239c9396debdbbb875b0d75d615c20308552/soc_zephyr.py#L85
- Checkout these files from master in `third_party/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog`.
-- `VexRiscv_Secure.v`
-- `VexRiscv_Secure.yaml`
-- `VexRiscv_SecureDebug.v`
-- `VexRiscv_SecureDebug.yaml`
- Checkout `core.py` from master in `third_party/litex/litex/soc/cores/cpu/vexriscv`.

However, this doesn't work with the `VexRiscv_SecureDebug` CPU variant. I get the following error from Vivado:
```
ERROR: [Synth 8-448] named port connection 'debugReset' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16438]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_address' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16439]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_data' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16440]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_wr' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16441]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_valid' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16442]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_ready' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16459]
ERROR: [Synth 8-448] named port connection 'debug_bus_rsp_data' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16460]
ERROR: [Synth 8-448] named port connection 'debug_resetOut' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/nebula/zephyr-on-litex-vexriscv/build/arty/gateware/arty.v:16461]
```

pawelsag, 2021-01-19:
Hi @lindemer
The issue should be fixed by 6665588. Could you verify if it works for you?

lindemer, 2021-01-19:
It builds with `secure+debug` now, thanks!

closed 2021-01-19 by lindemer