litex-hub/zephyr-on-litex-vexriscv #6: Generate Vivado hardware usage reports

lindemer, 2021-01-15:
I'd like to view the hardware use statistics after building this repository for an Artix-7 35T FPGA (e.g., https://github.com/SpinalHDL/VexRiscv/pull/148#issuecomment-722227413), but the report files don't appear to be generated automatically. Is there some way I can enable this? @kgugala maybe you know something about this? Thanks in advance.

kgugala, 2021-01-15:
Those are generated by default, you can find all the reports in `build/arty/gateware` directory. In e.g `arty_utilization_hierarchical_place.rpt` I got:

```
+----------------------------+---------------------------------+------------+------------+---------+------+------+--------+--------+------------+
|          Instance          |              Module             | Total LUTs | Logic LUTs | LUTRAMs | SRLs |  FFs | RAMB36 | RAMB18 | DSP Blocks |
+----------------------------+---------------------------------+------------+------------+---------+------+------+--------+--------+------------+
| arty                       |                           (top) |       4852 |       4666 |     184 |    2 | 3895 |     17 |     37 |          4 |
|   (arty)                   |                           (top) |       2677 |       2491 |     184 |    2 | 2655 |     16 |     27 |          0 |
|   FIFO_SYNC_MACRO          |                 FIFO_SYNC_MACRO |         23 |         23 |       0 |    0 |    0 |      0 |      1 |          0 |
|   FIFO_SYNC_MACRO_1        | FIFO_SYNC_MACRO__parameterized0 |         49 |         49 |       0 |    0 |    0 |      0 |      1 |          0 |
|   VexRiscv                 |                        VexRiscv |       2109 |       2109 |       0 |    0 | 1240 |      1 |      8 |          4 |
|     (VexRiscv)             |                        VexRiscv |        962 |        962 |       0 |    0 | 1016 |      0 |      2 |          4 |
|     IBusCachedPlugin_cache |                InstructionCache |        680 |        680 |       0 |    0 |  103 |      1 |      1 |          0 |
|     dataCache_1_           |                       DataCache |        482 |        482 |       0 |    0 |  121 |      0 |      5 |          0 |
+----------------------------+---------------------------------+------------+------------+---------+------+------+--------+--------+------------+

```

lindemer, 2021-01-16:
Thanks, sorry I have no idea how I missed that.

closed 2021-01-16 by lindemer